nakujaproject/N2-onboardTelemetryComputer #1: Review

0x6f736f646f, 2022-02-04:
Review schematic and PCB

- You can find the PCB parts list [here](http://nakujaproject.com/N2-onboardTelemetryComputer/production/index.html)
- The schematic is [here](http://nakujaproject.com/N2-onboardTelemetryComputer/schematic/Schematic.pdf)

SammyOina, 2022-02-04:
You seem to be missing reverse polarity protection as in [#10](https://github.com/nakujaproject/flight-control-computer/issues/10)

SammyOina, 2022-02-04:
consider adding another ws2812b led such that one indicated i2c communication to telemetry and another to flight control and possibly lora

SammyOina, 2022-02-04:
consider increasing the resistor values for the power and status led to ease on the heat produced by the led being too bright as was seen [here](https://github.com/QUALIS-LABS/ujuzi-pcb/blob/main/src/jlcpcb-fab/Schematic.pdf) @Mike-Kimani  can confirm this

SammyOina, 2022-02-04:
What is status 1 led used to indicate? If it multifunctional, consider an RGB led as it allows for many more possible indication options

SammyOina, 2022-02-04:
Is the flash memory circuit missing or will it be included in the pcb

0x6f736f646f, 2022-02-04:
> You seem to be missing reverse polarity protection as in [#10](https://github.com/nakujaproject/flight-control-computer/issues/10)

Done by https://github.com/nakujaproject/N2-onboardTelemetryComputer/commit/6289f43dfff52caed5843c49452a45dad5a00bb6

0x6f736f646f, 2022-02-04:
> consider adding another ws2812b led such that one indicated i2c communication to telemetry and another to flight control and possibly lora

Done by https://github.com/nakujaproject/N2-onboardTelemetryComputer/commit/6289f43dfff52caed5843c49452a45dad5a00bb6

0x6f736f646f, 2022-02-04:
> consider increasing the resistor values for the power and status led to ease on the heat produced by the led being too bright as was seen [here](https://github.com/QUALIS-LABS/ujuzi-pcb/blob/main/src/jlcpcb-fab/Schematic.pdf) @Mike-Kimani can confirm this

Done by https://github.com/nakujaproject/N2-onboardTelemetryComputer/commit/6289f43dfff52caed5843c49452a45dad5a00bb6

0x6f736f646f, 2022-02-04:
> consider increasing the resistor values for the power and status led to ease on the heat produced by the led being too bright as was seen [here](https://github.com/QUALIS-LABS/ujuzi-pcb/blob/main/src/jlcpcb-fab/Schematic.pdf) @Mike-Kimani can confirm this

It isn't public 😂

0x6f736f646f, 2022-02-04:
> What is status 1 led used to indicate? If it multifunctional, consider an RGB led as it allows for many more possible indication options

I have both RGB and white LED

0x6f736f646f, 2022-02-04:
> Is the flash memory circuit missing or will it be included in the pcb

Implemented by https://github.com/nakujaproject/N2-onboardTelemetryComputer/commit/6289f43dfff52caed5843c49452a45dad5a00bb6

closed 2022-03-22 by 0x6f736f646f