nakujaproject/N2-onboardTelemetryComputer #11: Review sd card circuit

0x6f736f646f, 2022-03-10:
## Prerequisites

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [x] I'm reporting the issue to the correct repository (for multi-repository projects)

## Context
Review the sd card circuit if it works
A simulation on proteus or multisim will be great

## What is the motivation/use case for changing the behaviour?
Store data

0x6f736f646f, 2022-03-10:
Review files can be found [here](https://github.com/nakujaproject/N2-onboardTelemetryComputer/pull/3/files)

0x6f736f646f, 2022-03-14:
@Simbona Kindly duplicate the same issue over at https://github.com/nakujaproject/internship2022/issues

Simbona, 2022-03-21:
The SD circuit interfacing with the MCU was properly done. SD card used in SPI mode with the MISO, MOSI, clock and chip select signals connected according to the SD Association specifications.

While the Proteus simulation wasn't conclusive (possibly due to bas SD card image format), I can assert the integrity of the schematic connections.
@0x6f736f646f

closed 2022-03-21 by Simbona