nakujaproject/N2-onboardTelemetryComputer #7: Review SDA lines for lora module and sd card

0x6f736f646f, 2022-03-10:
## Prerequisites

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [x] I'm reporting the issue to the correct repository (for multi-repository projects)

## Context
Review the SDA lines if they are correctly placed
- Do we need pull up or pull-down resistors?

## What is the motivation/use case for changing the behaviour?
To be able to send data to Lora and sd card

0x6f736f646f, 2022-03-10:
Review files can be found [here](https://github.com/nakujaproject/N2-onboardTelemetryComputer/pull/3/files)

0x6f736f646f, 2022-03-14:
@cmucheru and @Simbona Kindly duplicate the same issue over at https://github.com/nakujaproject/internship2022/issues

Simbona, 2022-03-21:
The SDA lines are well placed and compliant with best-practices. We expect the circuitry to work as indented.

Do we need pull up or pull-down resistors? We need pull-up resistors. That is, to prevent accidental writes to the SD card while the buses are not in use to prevent floating pins. 
@0x6f736f646f

closed 2022-03-21 by Simbona