nakujaproject/N2-onboardTelemetryComputer #8: Review reverse voltage protection circuit

0x6f736f646f, 2022-03-10:
## Prerequisites

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [x] I'm reporting the issue to the correct repository (for multi-repository projects)

## Context
Review the reverse voltage protection circuit if it works
A simulation on proteus or multisim will be great

## What is the motivation/use case for changing the behaviour?
Protect against reverse polarity

cmucheru, 2022-03-10:
> to @cmucheru <https://github.com/cmucheru>.
> <nakujaproject/N2-onboardTelemetryComputer/issue/8/issue_event/6218956408@
> github.com>

Thankyou for the information. I will review it and give the feedback.

0x6f736f646f, 2022-03-14:
@cmucheru Kindly duplicate the same issue over at https://github.com/nakujaproject/internship2022/issues

cmucheru, 2022-03-16:
Reviewed the circuit & found it to be working.
@0x6f736f646f

closed 2022-03-16 by cmucheru