nakujaproject/N2-onboardTelemetryComputer #9: Review voltage regulator circuit

0x6f736f646f, 2022-03-10:
## Prerequisites

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [x] I'm reporting the issue to the correct repository (for multi-repository projects)

## Context
Review the voltage regulator circuit if it works
A simulation on proteus or multisim will be great

## What is the motivation/use case for changing the behaviour?
Reduce 5V to 3.3V

0x6f736f646f, 2022-03-14:
@cmucheru and @Simbona Kindly duplicate the same issue over at https://github.com/nakujaproject/internship2022/issues

cmucheru, 2022-03-15:
The circuit is working in Proteus.
Observations:
- Some manuals recommended 10uF output capacitor for current stability.
@0x6f736f646f

closed 2022-03-15 by cmucheru

0x6f736f646f, 2022-03-15:
@cmucheru add this issue in the interns repo