nakujaproject/N2-onboardTelemetryComputer #14: Specific directories for each component inside the datasheet directory.

Mike-Kimani, 2022-04-05:
* **I'm submitting a ...**
  - [ ] feature request

Add specific directories to the datasheets directory. Currently, it's difficult to tell which datasheet belongs to which component.

Mike-Kimani, 2022-04-05:
[image]

How exactly? From here it's tedious to tell which datasheet belongs to the USB connector. Or to tell all these datasheets apart from one another.

0x6f736f646f, 2022-04-08:
Are you suggesting I add directories for each file?

Mike-Kimani, 2022-04-09:
Yes. Otherwise one has to open almost every datasheet to find a specific one which kind of beats the point of having a datasheets directory.

0x6f736f646f, 2022-04-09:
Done by https://github.com/nakujaproject/N2-onboardTelemetryComputer/commit/c50bcdbef7327d185672606a2ad6f20b47b54328

Mike-Kimani, 2022-04-16:
Cool.

closed 2022-04-16 by Mike-Kimani