nakujaproject/N2-onboardTelemetryComputer #15: Board and power camera power connectors.

Mike-Kimani, 2022-04-05:
* **I'm submitting a ...**
  - [ ] bug report

Change the board and camera power connections to match those in the N2 Power Distribution Board. [Here.](https://github.com/nakujaproject/n2_power_distribution_board/blob/main/schematic/Schematic_n2_power_distribution_board-png_2022-04-04.pdf)

Currently you have exactly the same connectors for +5V and +12V which should not be the case.

0x6f736f646f, 2022-04-08:
Why shouldn't it be the case?
I thought the board should have female connectors then we crimp the male connection

Mike-Kimani, 2022-04-09:
Safety. Having connectors with a different pitch for +12V and +5v rules out the possibility of anyone plugging in +12V into the +5v terminal.

0x6f736f646f, 2022-04-09:
Then why have I labelled it on the board?

Mike-Kimani, 2022-04-16:
> Then why have I labelled it on the board?

The labeling won't stop one from plugging in +12v into the +5v terminal if their connectors are exactly similar. We should ideally have connectors with different pitches but this may not be possible due to their unavailability locally.

We could have 3P for +12v connection with one more ground pin and 2P for the +5v connection. This way the +12V JST connector cannot possibly be plugged in to the +5V JST connector.

rodneyosodo, 2023-09-19:
Closing due to inactivity

closed 2023-09-19 by rodneyosodo