nakujaproject/N2-onboardTelemetryComputer #16: I2C connectors.

Mike-Kimani, 2022-04-05:
* **I'm submitting a ...**
  - [ ] bug report

Change I2C connectors from jumper wire header pins to JST connectors as currently on the avionics board. [Here](https://github.com/nakujaproject/avionicsv2/blob/reviews2/COMPUTER_PCB/Schematic_N2%20Avionics%20board.pdf)
 The pitch for the I2C connector is 1mm.

Jumper wire connections aren't firm enough.

0x6f736f646f, 2022-04-05:
The JST connectors we have available are compatible with 2.54mm header pins

@WashingtonKK Confirm on this

0x6f736f646f, 2022-04-08:
I'll stick with the header pins because:
- We already have the JST connectors here so we will just swap them if necessary
- The JST connectors we have it is tiresome to wire the female connections as it requires crimping that is why we changed from JST to header pins during the test flight
- They are quite expensive compared to the normal header pins

Mike-Kimani, 2022-04-09:
Regarding the comment above:

- You seem to still have a lot of space on the bottom side.
[image]

- We may have already been unable to acquire the data due to loose jumper wire connections. So it may already be too late to swap them.

-We can get already crimped cables.

- I don't think the difference in price is too much to ensure we don't loose data.

Mike-Kimani, 2022-04-09:
Actually our main problem is acquiring JSTs with different pitch sizes.

0x6f736f646f, 2022-04-09:
I have a lot of space at the bottom since JLCPCB will be assembling only one layer which is the top layer.

closed 2023-09-19 by rodneyosodo